PiSupply/iot-lora-gateway #24: Missing a module named "dialog"

shawaj, 2018-10-14:
Getting an error on fresh raspbian saying i am missing a module named "dialog"

```
pi@raspberrypi:~ $ iot-lora-configure
Traceback (most recent call last):
  File "/usr/local/bin/iot-lora-configure", line 4, in <module>
    from dialog import Dialog
ImportError: No module named 'dialog'
```

shawaj, 2018-10-14:
noticed that python3-dialog package should have been installed during the install. not sure why it wasn't. but after installing it manually it worked.

possibly just user error here but will try again on a fresh raspbian and close this if successful

ryanteck, 2018-10-14:
Indeed seems odd as it's the first thing installed by the installer. Will have to double check it.

shawaj, 2018-10-14:
Yeah maybe it was just a local issue. Perhaps WiFi was playing up or
something.

It's a weird one because I ran the installer the whole way through and it
completed ok without any errors but then got that.

ryanteck, 2018-10-20:
Was diagnosed as to be an issue with Wi-fi at the time of installation. However will try to add error handling in to alert the user so they know installation has failed.

ryanteck, 2018-10-30:
Closing this issue and creating a new one with the new issue.

closed 2018-10-30 by ryanteck