PiSupply/iot-lora-gateway #37: Gateway not working

ryanteck, 2018-12-02:
Two people (li0nic & @sbamueller) have reported issues with their gateways with the following issues:

Both have tested with different power supplys, checked the hats are mounted firmly etc.

The only thing left to diagnose is if its a hardware or software fault.

shawaj, 2018-12-03:
Is it worth us paying for these to be returned so we can diagnose easier?

ryanteck, 2018-12-03:
I think one may be getting returned already. The other I'm working on now as I've done a new SD card image which may solve the issue.

shawaj, 2018-12-03:
Ok great

ryanteck, 2018-12-17:
@sbamuller's gateway is now fixed and sent back to him. 

li0nics is on it's way back to us now. @shawaj

shawaj, 2018-12-17:
Awesome. Be interested to see if it's the same issue