PiSupply/iot-lora-gateway #49: support Libre Computer H2+/H3/H5

Tonymac32, 2019-07-16:
This should require a single entry, H2+/H3/H5 being pin compatible.

However, I noticed the board detection only pulls 1 word, on Armbian at least this can cause some real trouble, 

Libre computer
   H3
   H5
   Renegade

Tonymac32, 2019-07-16:
With an example I can just give you PR's for other boards, assuming the scripts can tell them apart.  I have NanoPi K1 Plus, Renegade, Pine64, Nanopi K2, Le Potato, etc.  (I'm an Armbian Dev)

ryanteck, 2019-07-18:
Hi @Tonymac32 

We did have tinkerboard support included I haven't tested for a while but don't think that stuff should have broke.

However realistically I don't think it's best to support all of these platforms, I think the included Tinker board support is fine and I'm happy to fix that. However many different boards (even if they are pin compatible) could cause issues. 

Realistically as well there could be support problems as for even just the ones you mentioned, there's different kernel versions and such along with some of the boards saying no official support from Armbian.

The main issue is while they're all pin compatible, they could all be on different SPI pinouts. For example the tinkerboard is on spidev2.0, rpi is on 0.0 and odroid C2 on 0.0. I'm pretty certain some SBCs i've used have been on /dev/spidev32766.0 or such.

Plus we'd also need to have a different file for each board for the GPIO Wiring, the K1 would be pin 3, le potato is 10 and tinkerboard on 178.

I mean it would be possible, however if I was to I would potentially need one of each board to confirm it works @shawaj . Plus it'd just be this basic installer and not the much better image and user interface provided by the RPi setup because to then support that as well would be significant.

shawaj, 2019-07-18:
For the amount of people that use those other platforms - I don't think it is really worth it. Additionally, I don't see how those platforms provide any real benefits over the Pi (other than, in some cases, being marginally cheaper). 

I guess the best thing would be for us to provide examples / basic instructions for getting stuff running / installed on other platforms and then some easy way of configuring it.

As you say @ryanteck we don't really need the support headache of supporting multiple platforms, when it is only a very small user base compared to Pi

ryanteck, 2019-07-18:
Could potentially document the lines of code that need changing for people to do it manually. As primarily this older software is just for those as a majority of the work has now gone into the SD Card images and soon APT Packaging for RPi and Standalone users.

I'll get tinkerboard fixed as that seems to be a fairly big community to support which is non Pi. However these others aren't enough to justify it.

closed 2019-07-18 by ryanteck

shawaj, 2019-07-18:
agreed - i think some basic documentation on what they could look at to change is enough

ryanteck, 2019-07-18:
Tinkerboard fixing is in progress, however multiple changes to Armbian has broken software supporting our reliability point of using an RPi as major changes such as board names and SPI Support are fairly reliable.

For best reliability in terms of software support, testing and what we design and test for the Raspberry Pi is the best. We're also working on making adjustments to enhance the reliability of SD Cards by switching to logging to ram files etc.

Tonymac32, 2019-07-18:
This is the first mention of an issue for Spidev, there have been no issues
raised with Armbian so it hadn't been spotted.  I will review/repair.

The aforementioned documentation would be most welcome, I can support a
fork of this for Armbian, and drop you a line if I think you'd welcome any
ideas, like for better board detection of there is any interest (following
mainline convention with device tree will always just return the vendor
name if I read your code correctly)

ryanteck, 2019-07-18:
To be honest the fact that there's been broken SPI and board information already just backs up the fact it is too much work to support Armbian boards. We've spent a significant amount of time perfecting and improving our Raspberry Pi Support and this is the main avenue we are going for support.

Tonymac32, 2019-07-18:
Unreported bugs don't get fixed, now it's been reported.

Board information was never broken.  Vendor kernel had the codename for the
platform, once the mainline support chose the Asus Tinker Board string I
changed the vendor kernel to match, which makes all Armbian images
consistent with each other and with mainline.

I am offering to help support, as it's obviously not welcome, I will be on
my way.